ArduPilot/ardupilot #8239: Feature Request - Copter - Direct Motor Passthrough

JacksonUAS, 2018-04-25:
#### Issue details
Currently we lack a direct throttle - motor output pass through mode for thorough power system testing of assembled aircraft.

Motor compass calibration appears to disable all mavlink telemetry, so not particularly useful for testing, it also appears to time out.

Motor Test doesnt allow use of the throttle stick, and has a 100 second limit. 

Could be setup as a mode, with warnings, or on a separate switch in the similar way to RTL/Motor Kill etc

#### Version
All

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

DavidIngraham, 2018-04-26:
This feature is actually implemented through the servo library. You can assign any RC input to any RC output.

The issue is that the copter code will re-assign the servo outputs to act as motor outputs if it detects that some of the required motor outputs do not have a servo output associated with them. 

On plane, these values are not reassigned, and you can use this method to test quadplane motor outputs.

Personally, I'm not a fan of re-assigning the parameters because it leads to potentially dangerous behavior that is unexpected. A user could believe that the servo outputs were disabled if they were unmapped, but the copter code would re-enable them automatically.

I believe the proper path forward is to disable the re-assignment and instead enable in arming check as described in #7354

DavidIngraham, 2018-04-26:
Also Discussed in #7152

DavidIngraham, 2018-04-26:
Actually, it looks like this has been fixed in master now, and should be working in 3.6. I will test and report back.

DavidIngraham, 2018-04-27:
Just confirmed that this works on Copter 3.6.

Set the SERVO_X function for whichever channel that you'd like to test to 53. That will make it output whatever value is coming in from the throttle channel (Assuming your throttle is on channel 3).
Please be careful when doing this. The motors will spin up even if the vehicle is not armed

closed 2018-04-27 by DavidIngraham

JacksonUAS, 2018-06-06:
This feature passes through at 50Hz, and ignores RC_SPEED, so isnt really suitable for power train testing as its sits.