ArduPilot/ardupilot #8258: Autotect MS5607 on px4fmu

kikislater, 2018-04-29:
#### Issue details
Some sellers deal some pixhawk px4fmu v2 with wrong baro. It's well known on rcgroups and they replace MS5611 by MS5607. It could be dangerous to fly with this wrong barometer.
We could watch after loading firmware on it that pressure is around 500 hpa or by disassembly it and watch 5607 written on barometer sensor. 
Watching PX4 code, I found they introduce an autodetect barometer. Could be nice to have the same on ardupilot for users that didn't realize wrong barometer is installed and prevent big crash injuries due to wrong altitude estimation.

More details on PX4 code here : http://discuss.px4.io/t/pixhawk-fmu-v2-3dr-pixhawk-1-and-ms5607-barometer/6490

#### Version
All versions

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
All frame

#### Hardware type
PX4FMU

#### Logs
No logs

jaxxzer, 2018-05-01:
I believe the problem with this is in flight reboots, but there is also a number of users flying with this baro. I suggest it is rather a parameter configuration. Do we have users that fly at this altitude? If so, we can have a statustext or flag at boot to indicate the pressure is not as expected, and have the gcs make further recommendation to check the hardware manufacturer, link to a wiki on the matter, or direct to the parameter to fix it.

kikislater, 2018-05-01:
How could it be dangerous to fly in tipycal altitude with this baro? What happen when planning a flight plan with this baro?
May be classic flights and zero behaviour but I will be afraid to fly with this barometer and have the wrong pressure.

jaxxzer, 2018-05-01:
> May be classic flights and zero behaviour but I will be afraid to fly with this barometer and have the wrong pressure.

The solution in the meantime is to buy hardware from a reputable source.

kikislater, 2018-05-01:
@jaxxzer I'm agree about sources but sometimes you could not depending on where you live and method of shipping/tax. But that's not the solution, there is some copy in the nature with this non conventional barometer on px4 fmu v2, some advised users know and others will never know and in my opinion it should be take on consideration. It's not only for me because I have others pixhawk and this one could be use for bench testing or not so far from the bin ;-) But for new users who buy from theses crap sellers would have a better experience with opensource autopilot...

jaxxzer, 2018-05-01:
@tridge are you aware of this problem? I know the 'pixhawk 2.4.6' come with ms5607, presumably the cost-optimized 2.4.8 does as well.

The ms5607 reports half the pressure (P/2) of the 5611 when driven with the current code.

OXINARF, 2018-06-20:
@rmackay9 We already support this baro, we just don't search for in the Pixhawk. So I changed the title.

@kikislater That detection approach seems very weak to me. AFAIK there is no reliable way to differentiate between MS56XX baro's.

tridge, 2018-06-20:
the first word in prom is "16 bit reserved for manufacturer". It is possible this includes part number information. We should ask Measurement Specialties

tridge, 2018-06-20:
I've sent a query to MS via their web form

hobbya, 2018-07-31:
Would it be feasible to make it user selectable on Mission Planner if MS has no response?

kikislater, 2018-08-01:
+1 if we could change it in parameters

acobo, 2018-08-02:
Hello to all,
 I have just purchased a cheap Pixhawk, noticed the huge error in pressure and altitude, and discovered that it has a MS5607 baro chip. I have checked the source code of ardupilot and,  as a workaround, I was considering to compile the sources and change just this line in MS5611.cpp

```
    case BARO_MS5611:
        _calculate_5611();
```
to 

```
    case BARO_MS5611:
        _calculate_5607();
```
which is already implemented. 

Should it work? Thank you in advance for your help.
Regards, Adolfo.

jaxxzer, 2018-12-20:
If we ever do 'solve' this, the same thing applies to the ms5837-30ba and ms5837-02ba. Right now we only support 30 bar version.

Gennady6414, 2019-03-10:
Hello all! 
I bought Pixhawk PX4 2.4.8 and at the start it reports that it has no baro. Is there anything I can do to work around this problem? Is it ok to fly without the baro? Since I am new to GitHub, please, send a copy of your response to my e-mail: TommyJoad2017@gmail.com

Jaaaky, 2019-05-02:
Any update on this?

I noticed that values set in hwdef.dat are not respected. Changed this in fmuv3/hwdef.dat to ms5607 but nothing changed.
We can add a parameter like GND_EXT_PROBE for internal ones and force the user selection?

pellhall, 2020-05-21:
Is there a way to add an external MS5611 on the I2C port and disable the internal one? Or has this issue been addressed in the code already? Not sure if my soldering skills are up to R&R of that pad mounted little $10 guy.

ES-Alexander, 2023-02-17:
@tridge was this effectively fixed in 13893d3? And if so could a similar approach be used for distinguishing `MS5837-30BA` (current default) from `MS5837-02BA` (currently not supported), as @jaxxzer [requested above](https://github.com/ArduPilot/ardupilot/issues/8258#issuecomment-449008302)?

@Williangalvani resolving this would add Bar02 support to Sub

IAMHAADICOOL, 2023-10-15:
Hey, was this issue resolved? Can anyone help me on how to compile ardusub firmware to enable support for bar02 pressure sensor?

IAMHAADICOOL, 2023-10-15:
> @tridge was this effectively fixed in [13893d3](https://github.com/ArduPilot/ardupilot/commit/13893d3ed57ee6f250c135395d08ecd08ac77342)? And if so could a similar approach be used for distinguishing [...]

Can you please provide an update on how to resolve this issue? How to build custom firmware with support for MS5837-02BA?

IAMHAADICOOL, 2023-10-18:
> @tridge was this effectively fixed in [13893d3](https://github.com/ArduPilot/ardupilot/commit/13893d3ed57ee6f250c135395d08ecd08ac77342)? And if so could a similar approach be used for distinguishing [...]

Hey, was this fixed? If so, what were changes made to the script? Can you please tell?

IAMHAADICOOL, 2023-10-18:
> I've sent a query to MS via their web form

Can you please tell the changes and the corresponding scripts to which they were made to enable the support for  MS5837-02BA?

peterbarker, 2024-07-31:
> @tridge was this effectively fixed in [13893d3](https://github.com/ArduPilot/ardupilot/

Yes, that was worked around with that patch.

I'm closing this issue now; if we need to do something with MS5837 please open a new issue.

We'd *REALLY* rather not add more hacks as we did for the MS5607, so if you can find a way to distinguish with registers that is *far* preferable.