ArduPilot/ardupilot #8272: Triggering camera from MP permanently sets feedback pin to 3.3.V

Kazcade, 2018-05-01:
#### Issue details
Triggering the camera relay from MP via DO ACTION -> Trigger Camera Now sets the camera feedback pin (if configured) to 3.3V until the FC is power cycled. This happens regardless of CAM_FEEDBACK_PIN used or the selected polarity (CAM_FEEDBACK_POL).

This makes bench testing impossible, since the feedback pin can then only be triggered by connecting it to GND. Can anybody confirm this? I've found somebody reporting a similar issue in 2016 (https://discuss.ardupilot.org/t/cam-feedback-pin-output-3-3v-when-it-should-not/13613), but i've found no follow-up on this.

#### Version
Plane 3.8.5

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Hardware type
Pixhawk 2.1

Kazcade, 2018-05-02:
As far as i can tell from looking at AP_Camera.cpp, the feedback pin is apparently set to 3.3V regardless of CAM_FEEDBACK_POL during the feedback check:
321 // enable pullup
322    hal.gpio->write(dpin, 1);

Is this intended?

geofrancis, 2018-06-05:
yes since the hotshoe is essentially a switch shorting it to ground.

Kazcade, 2018-06-06:
CAM_FEEDBACK_POL defines if feedback is triggered when the pin is held high or when it is held low. How do i hold the feedback pin high when it is set to 3.3 V regardless of CAM_FEEDBACK_POL?

Kazcade, 2018-06-07:
Whatever it was, the polarity setting is working now and after additional testing i can reliably trigger feedback by holding the pin high or low. Closing this issue, as i can probably ascribe this to user error. Thanks for replying.

closed 2018-06-07 by Kazcade