ArduPilot/ardupilot #8278: px4 toolchain git repository fatal

emretuncay, 2018-05-02:
#### Issue details
I am trying to custom octo copter using ardupilot on pixhawk.I installed px4 toolchain and I changed mixfile based on my octocopter.When I make px4-v2 I get some fatal like below
[image]
How can I solve this fatal.Can you help you.
Thank you

#### Version

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
octa

#### Hardware type
pixhawk

#### Logs

amilcarlucas, 2018-05-02:
You did not read the instructions :)
Please issue a `git submodule update --init --recursive` in the root directory of ardupilot

closed 2018-05-02 by amilcarlucas

khancyr, 2018-05-02:
Hello,
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

emretuncay, 2018-05-02:
Thank you amilcarlucas 
I read the instructions and submodule worked 🥇