ArduPilot/ardupilot #8288: Support for Eagle tree airspeed sensor v3

Karthikaero, 2018-05-03:
#### Issue details
_It would be great to see support for Eagle tree Airspeed sensor v3 in Arduplane. My use requires higher velocity ranges which is supported by the ETS sensor only._

#### Version
_I have tested with AP3.8.2 with no luck_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_Fixed wing_

#### Hardware type
_Pixhawk 2.1 cube_

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2018-05-04:
Can you link the data sheet with the protocol?

auturgy, 2018-05-04:
We used to have a driver in the px4firmware tree. Might still be there.

Regards,

James

Karthikaero, 2018-05-04:
[microsensor-i2c.pdf](https://github.com/ArduPilot/ardupilot/files/1973180/microsensor-i2c.pdf)
@magicrub Here we go!!!!

marcdornan, 2018-11-03:
Any update on this? I have a few of these sensors that I would like to use?

peterbarker, 2020-12-04:
I've created a PR for a driver for this sensor ( #15971).

However, we do need to know if it is suitable for providing data to the vehicle for flight.

@marcdornan @Karthikaero are you in a position to test this new driver as a ride-along device with another sensor?

OpenUAS, 2021-01-22:
Used successfully in flights with Paparazzi UAS, we used tons of them, but nowadays there are better alternatives. It was a nice pull request, I feel the pain of not honoring the PR for the developer involved, but understand the logic behind the sad choice. That being said, the code change is not so big, one might as well have honored the pull request.

peterbarker, 2021-01-23:
Thanks @OpenUAS - it was partly me trying to close old issues, partly me exercising the i2c simulation framework.

Both seem to have worked.

I guess I'll close this PR with a "apparently WONTFIX" :-)

closed 2021-01-23 by peterbarker

Dajtek, 2023-10-23:
Have anybody schematic because I need C14 capacity, my was broken.