ArduPilot/ardupilot #8292: Mavlink OSD throttle always at 50% with twin engine setup

kingdraki, 2018-05-03:
#### Issue details
With twin engine setup, 
SERVOn_FUNCTION = 73, 
SERVOn_FUNCTION = 74, 
throttle reported at Mavlink OSD is always 50% when no SERVOn_FUNCTION = 70 is present at setup.
When i put SERVOn_FUNCTION = 70 on an unused servo port OSD is working again.

#### Version
3.8.4 & 3.8.5

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Twin Engine Plane

#### Hardware type
Pixhawk

#### Logs
PARAM File:
[20180501_ardupilot_384.zip](https://github.com/ArduPilot/ardupilot/files/1972426/20180501_ardupilot_384.zip)

yak-54, 2018-05-04:
I have the same issue my twin engine planes  3.8.4 & 3.8.5

PARAM File:

[rycbo-19-4-18-22-41-32.zip](https://github.com/ArduPilot/ardupilot/files/1976145/rycbo-19-4-18-22-41-32.zip)

IamPete1, 2021-08-15:
I'm not really sure what we can do about this in AP, it sounds like a the OSD id just looking at the wrong thing? Maybe it is fixed now?

IamPete1, 2021-08-30:
This is a issue with the MAVLink OSD. Happy to work with a OSD dev to get this fixed, but as I said, I'm not sure what we can do in AP to resolve this.

closed 2021-08-30 by IamPete1

yak-54, 2021-08-31:
its been fixed in 4+

IamPete1, 2021-08-31:
Well, I'm surprised.