ArduPilot/ardupilot #8304: This file is invalid when trying to load from MP

lordneeko, 2018-05-04:
https://github.com/ArduPilot/ardupilot/blob/4e53d9e1a836a388e0fe947468772912b9016e1e/Tools/Frame_params/SkyViper-2450GPS/defaults.parm#L1

I loaded AC 3.6 dev-1 to my Skyviper and wanted to try to get it working by loading this default.parm

But when I try to load it from MP I get the following error. Also, the skyviper default params don't appear to be compiled into the ardupilot/ardupilot master. If i reload from the firmware.skyviper.com all my parameters get set back up correctly and it will fly.  With the master branch..everything is set to default and the craft won't fly.

[image]

meee1, 2018-05-05:
the file you downloaded is corrupt/ a html page.

please redownload and don't use save target as.

lordneeko, 2018-05-07:
Thanks. I'll close this then.

closed 2018-05-07 by lordneeko