ArduPilot/ardupilot #8310: Plane: airbrakes using dual rudder mixing

magicrub, 2018-05-06:
See example F-15 Tomcat where the rear dual rudders could be used for airbrakes
https://www.youtube.com/watch?v=7rvGGwGEUXU&feature=youtu.be&t=3054

Here's the FB post
https://www.facebook.com/groups/ArduPilot.org/permalink/1812223075467544/

serchist, 2018-05-06:
Differential spoilers also can be used as airbrakes.
Autopilot can make use of airbrakes when it needs to slow down.
As I see, say, AIRBR_CHAN parameter should be added for input channel and a few new values for SERVOn_FUNCTION: airbrake itself, left rudder, right rudder.  And, may be, a parameter for mixing airbrake into differential spoilers.

Naterater, 2019-01-05:
I think this should be closed in favor of https://github.com/ArduPilot/ardupilot/issues/6375. Mixing surfaces to conduct air braking would be awesome. What do you think @magicrub?

peterbarker, 2020-12-04:
This could now be done in terms of the merged airbrake support, based on `airbrake_update` and probably present in `set_servos`.

Another parameter would be required, probably some scaling factor from `airbrake_pc` to an addition to the `SRV_Channel::k_rudder`