ArduPilot/ardupilot #8314: Servo gimbal not working if ESCType OneShot125

SergeyBokhantsev, 2018-05-06:
#### Issue details
Having camera tilt servo function on out #5
If in MP set ESC Type = OneShot125 => camera servo goes crazy (moves out of limits)
If set ESC Type = Normal => gimbal servo works ok.

#### Version
3.6-rc0, Nuttx

#### Platform
[x] Copter

#### Airframe type
Quad copter X

#### Hardware type
Pixracer

SergeyBokhantsev, 2018-05-06:
also, here is how MP ServoOutput screen looks like with Normal ESC type:
[image]

And here how its looks with OneShot125:
[image]

OXINARF, 2019-06-11:
@SergeyBokhantsev Does this also happen with ChibiOS?

SergeyBokhantsev, 2019-06-11:
@OXINARF no issue with ChibiOS, fixed :)

closed 2019-06-11 by OXINARF