ArduPilot/ardupilot #8318: Radio channels 1-4 dead after connecting to QGroundControl under Linux

mkunzBGC, 2018-05-07:
I posted this first [on the qgroundcontrol issue tracker](https://github.com/mavlink/qgroundcontrol/issues/6370), but according to @DonLakeFlyer  it is actually a bug in ArduCopter.

With the following system

- Ground station operating system: Ubuntu 16.04.1, kernel 4.13.0-38-generic
- QGroundControl 3.3.2 downloaded from here
- Autopilot board: Pixhawk 2 The Cube
- Autopilot firmware: ArduPilot:Copter 3.5.5
- RC transmitter: FrSky Taranis running OpenTx 2.2.1, XJT module runs XJT_EU_V20_build170317.hex.frk
- RC receiver: X8R running X8R_EU_build1702102.frk
- RC receiver and transmitter are bound in D16 mode, channels 1 - 8 are transmitted.

starting QGC seems to disturb the autopilot such that the RC channels 1 through 4 are no longer received.

**This is how I can reliably reproduce the problem:**

1. Turn on transmitter and Pixhawk, wait until startup complete
2. Try to arm (throttle min, yaw right for 2 s) -> buzzer indicates that arming is not possible (no GPS because I am in the basement). This tells me that RC channels 3 (throttle) and 4 (yaw) are being received correctly by the autopilot.
3. Start QGC via USB cable, wait until parameters have been received
4. Under Radio Setup - Channel Monitor, channels 1 through 4 do not react on stick inputs (1 and 4 are centered, 2 and 3 are constantly at minimum). HOWEVER, channels 5 and 6 still work correctly!
5. When trying to arm now, the buzzer stays quiet and there is no message on the QGC screen.
6. Close QGC, confirm vehicle disconnect in the window that's popping up.
7. When trying to arm now, the buzzer is still quiet.

I have to restart the autopilot to make it respond to the arming attempt again.

**This problem also occurs when...**

- using telemetry radios instead of an USB cable between ground station and autopilot or
- running ArduPilot:Copter 3.6-dev or
- running QGroundControl 3.3.0

**This problem does not occur when...**

- running QGroundControl 3.3.2 on Windows 10 in a virtual box on the very same laptop or
- running Mission Planner 1.3.52 under Ubuntu on the very same laptop using Mono or
- running ArduPilot:Copter 3.4.6 (but this has other problems in the interplay with QGC, e.g. missing parameters)

At the moment I fly using QGC on the virtual Win10 machine, but would very much prefer do use the GNU/Linux version. I am happy to provide more input in order to resolve this issue.

@DonLakeFlyer has looked into this problem as described [here](https://github.com/mavlink/qgroundcontrol/issues/6370#issuecomment-385531071).

DonLakeFlyer, 2018-05-07:
The question is what stops it from sending channels 1-4?

WickedShell, 2018-05-07:
Most probable thing I can think of is if MANUAL_CONTROL messages are being sent from the GCS. @mkunzBGC  do you have a TLOG?

DonLakeFlyer, 2018-05-07:
Hmm, I had testing this myself as well. Let me double-check that I didn't have joystick on while testing.

DonLakeFlyer, 2018-05-07:
Yup. I think I had joystick accidentally turn on while testing. QGC with PX4 firmware has a test which disables radio cal when joystick is used. I need to figure out how to do the same thing with ArduPilot.

@mkunzBGC Was joystick your problem as well?

mkunzBGC, 2018-05-08:
Yes, indeed! Unchecking "Virtual joystick" in QGroundControl's "General" settings fixed made radio channels 1-4 behave correctly.
Can a test similar to what QGC and PX4 do be implemented for QGC and ArduPilot in order to prevent this user mistake?

closed 2018-05-08 by mkunzBGC

mkunzBGC, 2018-05-09:
I forgot the most important point: Thanks @WickedShell and @DonLakeFlyer for looking into this!