ArduPilot/ardupilot #8323: Documentation of how to calculate ATC_STR_RAT_FF for Rover

mikerob, 2018-05-08:
#### Issue details
In older versions, STEER2SRV_P was based (at least initially) on turn radius.   The instructions for ATC_STR_RAT_FF do not include a starting point. 

https://github.com/ArduPilot/ardupilot/blob/master/libraries/APM_Control/AP_SteerController.cpp includes the following comment at line 81: "The feed forward gain for steering this is the ratio of the achieved turn rate to applied steering. A value of 1 means that the vehicle would yaw at a rate of 45 degrees per second with full steering deflection at 1m/s ground speed."

Is this actually the correct definition of ATC_STR_RAT_FF for Rover?

I tried to work out the maths myself and got that the FF should be turn_radius x pi/4 - is this correct?  A second example for a value of 2 would help clarify whats on the numerator and what's on the denominator.

Could this be added to the documentation? 

#### Version
>3.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

#### Airframe type
RC car

#### Hardware type
Cube

#### Logs
N/A

rmackay9, 2018-05-08:
@mikerob,
The info you've dug out of AP_SteeringControl is only used for the ground steering on Plane.  Rover uses the APM_Control/AR_AttitudeControl library.  I guess you've seen the wiki instructions [here](http://ardupilot.org/rover/docs/rover-tuning-steering-rate.html)?

I don't know of a formula to calculate what the FF should be.  I always just plot the desired vs achieved turn rate while in Acro mode and then increase the ATC_STR_FF until the achived matches the desired fairly well.

I'm going to close this issue if that's OK.  requests for documentation improvements should really go into the [wiki issues list](https://github.com/ArduPilot/ardupilot_wiki/issues).

closed 2018-05-08 by rmackay9

mikerob, 2018-05-08:
Ack, I've seen those instructions. Hadn't realised I'd picked up the wrong code section.  Will try your approach.