ArduPilot/ardupilot #8324: Support setting custom parameters used by companion computers

envirover, 2018-05-08:
#### Issue details
Companion computers extend functionality of the autopilot in various ways and on various platforms and often have their own custom configuration settings not included in the set of on-board parameters supported by the ArduPilot profiles. 

For example, I'm developing a companion computer that runs on Raspberry Pi and requires a server hostname and access credentials as configuration parameters. The user experience for setting these parameters is not very nice. It requires SSH to the Raspberry Pi, updating a config file, and making the OS read-only to protect from SD card corruption.

On the other hand ArduPilot provides a very nice way for setting on-board parameters in GUI GCS that is very familiar to all the users.  It would be great if users could set custom on-board parameters of companion computers using the same workflow.

#### Version
All

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
All

#### Hardware type
All

#### Logs
All

rmackay9, 2018-05-08:
I've thought about this feature as well in the past.  It's a great idea for sure.  It's probably technically not an ArduPilot feature though, it's more likely something that could be added to dronekit or APSync.

peterbarker, 2018-05-08:
On Mon, 7 May 2018, Envirover wrote:

> On the other hand ArduPilot provides a very nice way for setting on-board
> parameters in GUI GCS that is very familiar to all the users. It would be
> great if users could set custom on-board parameters of companion computers
> using the same workflow.

With a bit of work GCSs should be able to target the companion computers 
to set their parameters.  The CC just need to take parameter 
setting/getting via mavlink.

A very small example of this is the esp8266 WiFi-serial adapter, which QGC 
can deal with (and MAVProxy, kinda, with a branch of mine).

envirover, 2018-05-16:
Does MAVLink itself support setting custom (not yet enumerated) parameters?

"MAV_CMD_DO_SET_PARAMETER - Set a system parameter. Caution! Use of this command requires knowledge of the numeric enumeration value of the parameter."

OXINARF, 2018-11-26:
Closing this as it isn't an ArduPilot autopilot feature. Several companies already take advantage of MAVLink to have parameters on companion computers.

closed 2018-11-26 by OXINARF