ArduPilot/ardupilot #8349: Reinstate setting of SYSID_THISMAV as part of sim_vehicle.py

peterbarker, 2018-05-10:
#### Issue details

We had to revert https://github.com/ArduPilot/ardupilot/commit/f8464577ac3ac0c22c2fbb9eb5011f23bd3bc8d6 as it broke standard invocations of SITL.

#### Hardware type
SITL

#### Logs
```
rancisco, 21:36
what was the issue? instance 0?

21:36
I tested everything except running without parameters...
Yeah, instance 0.
Francisco, 21:36
running without parameters shouldn't be a problem actually

21:36
It was :-)
Francisco, 21:37
how?

sim_vehicle sends instance 0 by default

21:37
BTW, you can't actually use "instance 0" as such - setting your SYSID_THISMAV to that value will have Not Good Consequences.
Francisco, 21:37
but launching SITL without parameters should never go through that code

21:37
I added a sanity check that id was greater than zero :-)
Francisco, 21:38
yeah, but only setting an instance number

if you don't give that to SITL then that check doesn't run

so I can see that default sim_vehicle run gives issue, but not running SITL without parameters

21:39
Ah, sorry, I was refering to sim_vehicle.py
Francisco, 21:39
still I think it would have been better to just change SYSID_THISMAV to be set to instance+1, and change the check to 0-254

21:39
Mmmm.
I think ti would be nice to have ./sim_vehicle.py -I 37 set the vehicle ID to 37
It means we have to default -I to 1
... and probably bind port numbers 5760+(n-1)*10
Francisco, 21:40
that works too

21:41
I was going to poke Pierre on it.  Should we open an issue to record these thoughts?
Francisco, 21:41
yeah probably

```

khancyr, 2018-05-11:
hummm I don't really know what the best behavior here : 
- hardcore instance 0 ID to 1
- default -I to 1 and shift all port with 5760+(n-1)*10
- change SYSID_THISMAV to be set to instance+1

I am balanced between the first and second. The first is the simplier and quite nobody should notice it. The second it the more correct one but could break those who are already using multiple vehicles ... or not as they should use a trick to manually assign ID ... so that should be good .

peterbarker, 2020-12-02:
We've got new and improved, better ways of doing sysid_thismav in master.

closed 2020-12-02 by peterbarker