ArduPilot/ardupilot #8353: SITL fail to compile with GCC 8.1

patrickelectric, 2018-05-10:
#### Issue details
SITL fail to compile with GCC 8.1
Works fine with clang 6.0.0

#### Version
5b56887428bd1cf300cbc9d2bc17cd1c6c1fb67c

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Logs
```
In function ‘void mav_array_memcpy(void*, const void*, size_t)’,
    inlined from ‘void mavlink_msg_named_value_float_send(mavlink_channel_t, uint32_t, const char*, float)’ at libraries/GCS_MAVLink/include/mavlink/v2.0/ardupilotmega/../common/./mavlink_msg_named_value_float.h:158:21,
    inlined from ‘void AP_Airspeed::read()’ at ../../libraries/AP_Airspeed/AP_Airspeed.cpp:404:43:
libraries/GCS_MAVLink/include/mavlink/v2.0/ardupilotmega/../protocol.h:176:9: error: ‘void* memcpy(void*, const void*, size_t)’ forming offset [5, 10] is out of the bounds [0, 4] [-Werror=array-bounds]
   memcpy(dest, src, n);
   ~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

```
This issue was erroneous opened in mavlink https://github.com/mavlink/mavlink/issues/903.

OXINARF, 2018-05-10:
From me in Gitter:

> That's a hard one, that's actually an issue in our code (https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Airspeed/AP_Airspeed.cpp#L404) (and it might happen elsewhere). So the issue is that where it is putting a string AS2, it should actually be sending a pointer to an array with size 10

magicrub, 2018-05-10:
yeah, really needs to be fixed in the mavlink generator. A temporary workaround in the ardupilto source could be something like: 
`mavlink_msg_named_value_float_send(MAVLINK_COMM_0, AP_HAL::millis(), "AS2", get_airspeed(1));`

to 
`const char str[MAVLINK_MSG_NAMED_VALUE_FLOAT_FIELD_NAME_LEN] = {'A','S','2',0,0,0,0,0,0,0];
mavlink_msg_named_value_float_send(MAVLINK_COMM_0, AP_HAL::millis(), str, get_airspeed(1));`

It's great that the compile caught this.

OXINARF, 2018-05-10:
@magicrub That's quite debatable. There is no such thing as strings in MAVLink; there are arrays of char, some of which even explicitly say they don't have to be null-terminated if they fill the entire array. Quite likely we can special case arrays of char in the generator but I'm not sure we want to do that, it gets messy really easy.

The change you propose isn't a workaround, it's the right thing to do, at least with the current generator.

closed 2018-05-16 by OXINARF

OXINARF, 2018-05-18:
@patrickelectric that's wrong though. MAVLink expects an array of length 10, independently of compiler version. The fact it hasn't given any issues so far is just luck.

PR #8383 (2018-05-16): Use named float wrapper to fix GCC 8.1 compile error
Fixes #8353 with a silly 10 byte name. Required to let GCC 8.1 work.