ArduPilot/ardupilot #8384: copter crash detection disarms copter and hence stops logging

GekoCH, 2018-05-13:
#### Issue details
When a Copter detects a crash while in air it releases the parachute and disarms the motors. However at the same time the logging will als be disabled. For further analysis this is not the best way. Of course you could set the parameter to always log even if disarmed but it will generate huge log files. 

So could we add a line of code where it says: If crash detect do not stop logging till hiting the ground.

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2018-05-14:
I agree this is a good idea.

peterbarker, 2018-05-14:
On Sun, 13 May 2018, Randy Mackay wrote:

> I agree this is a good idea.

There's a few ways this could be done.

One option would be for the cash detection code to do a 
set-but-not-a-set-and-save on the LOG_DISARMED parameter.

We could also have a method that set a state variable directly in 
DataFlash.

IamPete1, 2021-08-30:
Closed by https://github.com/ArduPilot/ardupilot/pull/8836

closed 2021-08-30 by IamPete1