ArduPilot/ardupilot #8390: Copter: WPNAV_SPEED value is changed by switching to RTL mode

horiuchi4dn, 2018-05-14:
#### Issue details
If RTL_SPEED is set to a value other than 0, switching to RTL mode changes the WPNAV_SPEED to the RLT_SPEED value, and the speed of subsequent missions is affected by that value.

This behavior causes surprises when resuming the mission from the RTL state.

#### Version
Copter 3.5.5
Copter 3.6-dev

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

khancyr, 2018-10-24:
Hello, 

I confirm the bug. It can be easily reproduce on SITL. Set a mission, set RTL_SPEED to 250 , launch the mission (the default speed is 500) , change mode to RTL, wait 5s, get back to auto, the mission will continue but at 250cm/s

IamPete1, 2021-08-30:
@rmackay9 's fix has been merged, so I guess this issue is resolved.

closed 2021-08-30 by IamPete1