ArduPilot/ardupilot #8404: Rover: add PIVOT_TURN_RATE parameter to control max rotation rate of pivot turns

rmackay9, 2018-05-16:
This issue was raised [here in the discuss forums](https://discuss.ardupilot.org/t/skid-steer-mower-overshooting-pivot-turns/28910/18).

In Rover-3.3.1 the vehicle attempts to do pivot turns at the maximum rotation speed of the vehicles (held in the ATC_STR_RAT_MAX parameter) which is too high for some users.  We should add a new PIVOT_TURN_RATE parameter which holds the desired rate in degrees/second.  We could set the default to some reasonable value (90deg/sec perhaps?) or we could default it to zero in which case the vehicle would use the ATC_STR_RAT_MAX parameter as the rotation rate.

rahulgo, 2018-05-17:
Hi I'm new to Open Source development. Would this be something I can tackle?
Thanks

rmackay9, 2018-05-17:
@rahulgo, you can definitely give it a try!

closed 2018-05-21 by rmackay9