ArduPilot/ardupilot #8416: Compiled Ardupilot Copter from Master will not run on Intel Aero RTF

trgiman, 2018-05-16:
#### Issue details
_Please describe the problem, or desired feature_

Compiled source will not run on Intel Aero RTF FC.Compiled src is uploaded but is not running. Precompiled ardupilot ver 3.5.3 delivered by default on Intel Aero Board works. But I need to change some code in Ardupilot Copter so i need to compile my own.

#### Version
_What version was the issue encountered with_
Master branch

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

Intel Aero RTF

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

Intel Aero RTF AscTech v2

#### Logs

./waf configure --board aerofc-v1 

Setting top to                           : /home/user1/Ardupilot/ardupilot 
Setting out to                           : /home/user1/Ardupilot/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : aerofc-v1 
Checking for program 'arm-none-eabi-ar'  : /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-ar 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-gcc 
Checking for HAVE_CMATH_ISFINITE         : no 
Checking for HAVE_CMATH_ISINF            : no 
Checking for HAVE_CMATH_ISNAN            : no 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : no 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no 
Checking for header endian.h                   : not found 
Checking for header byteswap.h                 : not found 
Checking for program 'cmake'                   : /usr/bin/cmake 
Checking cmake version                         : 3.5.1 
Checking for program 'ninja, ninja-build'      : /usr/bin/ninja 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.12 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.12 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Gtest                                          : STM32 boards currently don't support compiling gtest 
Checking for program 'arm-none-eabi-size'      : /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-size 
Benchmarks                                     : disabled 
Unit tests                                     : disabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (0.630s)

Than I run command

 ./waf copter --upload

Waf: Entering directory `/home/user1/Ardupilot/ardupilot/build/aerofc-v1'

.....

which produce several warnings including:

Configuring NuttX for aerofc-v1
Exporting NuttX for aerofc-v1
sem_timedwait.c: In function 'sem_timedwait':
sem_timedwait.c:226:23: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
   if (abstime->tv_sec < 0 || abstime->tv_nsec > 1000000000)
                       ^
timer_settime.c:70:1: warning: 'inline' is not at beginning of declaration [-Wold-style-declaration]
 static void inline timer_sigqueue(FAR struct posix_timer_s *timer);

Resulting binary is uploaded to AeroFC but it is not working.Do you have got any suggestions?

lucasdemarchi, 2018-05-17:
Hi @trgiman 

I can help, but I don't have a board with aerofc handy anymore. If we don't find the reason why it's not working I may try to find one.

Please log in the aero board and post the output of the following commands:

```bash
aero-get-version.py
systemctl stop mavlink-router
cat /etc/mavlink-router/main.conf
mavlink-routerd
```

For the last command, let it cycle through the baud rates twice.

trgiman, 2018-05-17:
Hi @lucasdemarchi,

Thanks for help. Below is the output of commands. FYI. I have no problem to deploy and use standard Ardupilot (delivered as part of Yocto, but I need to compile my own since I did changes in the code) and also have no problems to deploy and use my custom builds  of PX4 firmware.

 **aero-get-version.py:**
 
BIOS_VERSION = Aero-01.00.13
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.6.1 (pyro)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc2
AeroFC firmware version = unknown (do not know why but for custom builds it always show unknown version)

**cat /etc/mavlink-router/main.conf**

[General]
TcpServerPort=0
ReportStats=false

[UartEndpoint uart]
Device = /dev/ttyS1
Baud = 921600

[UdpEndpoint ethernet-over-usb]
Mode = Normal
Address = 192.168.7.255
Port = 14560

[UdpEndpoint wifi]
Mode = Normal
Address = 192.168.8.255
Port = 14560

**mavlink-routerd**

Open UDP [4] 192.168.7.255:14560  
Open UDP [5] 192.168.8.255:14560  
Open UART [6] /dev/ttyS1 *
UART [6] speed = 921600

Initially I though that  I used wrong version of cross compiler. But I am using correct one according the Ardupilot documentation -  gcc-arm-none-eabi-4_9-2015q3. So compiled code is flashed to Aero Flight Controller. But than nothing. However during compilation I get various Nuttx warnings.

khancyr, 2018-05-17:
Could you try with ardupilot master or 3.5 branch instead of your fork?

trgiman, 2018-05-18:
@lucasdemarchi I added the baudrate. However result is only endless cycling in between of the baudrates.

Open UDP [4] 192.168.7.255:14560  
Open UDP [5] 192.168.8.255:14560  
Open UART [6] /dev/ttyS1 *
UART [6] speed = 460800
Retrying UART [6] on new baudrate: 921600
UART [6] speed = 921600
Retrying UART [6] on new baudrate: 460800
UART [6] speed = 460800
Retrying UART [6] on new baudrate: 921600
UART [6] speed = 921600
Retrying UART [6] on new baudrate: 460800
UART [6] speed = 460800
Retrying UART [6] on new baudrate: 921600
UART [6] speed = 921600
Retrying UART [6] on new baudrate: 460800
UART [6] speed = 460800
Retrying UART [6] on new baudrate: 921600
UART [6] speed = 921600
Retrying UART [6] on new baudrate: 460800
UART [6] speed = 460800
Retrying UART [6] on new baudrate: 921600
UART [6] speed = 921600
Retrying UART [6] on new baudrate: 460800
UART [6] speed = 460800
Retrying UART [6] on new baudrate: 921600
UART [6] speed = 921600
Retrying UART [6] on new baudrate: 460800
UART [6] speed = 460800
Retrying UART [6] on new baudrate: 921600
UART [6] speed = 921600
Retrying UART [6] on new baudrate: 460800
UART [6] speed = 460800
Retrying UART [6] on new baudrate: 921600
UART [6] speed = 921600
Retrying UART [6] on new baudrate: 460800
UART [6] speed = 460800
Retrying UART [6] on new baudrate: 921600
UART [6] speed = 921600
Retrying UART [6] on new baudrate: 460800

@khancyr 

my fork is master forked from yesterday. I was not able to compile 3.5 branch but I will try it again and will update it here. Never the less I have read blogs where @rmackay9  mentioned, that 3.6 version is right one for Aero RTF, never the less yocto image delivered by intel has 3.5.3 version compiled (which I am able to deploy and use).

trgiman, 2018-05-19:
@lucasdemarchi  @khancyr I was able to compile 3.5.5 and run 3.5.5 ArduPilot Copter version (my fault, originally I used wrong branch updates. Now compilation of 3.5.5 is without warnings and errors). But still have no success with 3.6 which is recommended one

magicrub, 2018-05-21:
@lucasdemarchi would you like me to send you my aero?

PR #8609 (2018-06-28): Fix panic() on aerofc
This should fix #8416
closed 2018-06-28 by lucasdemarchi