ArduPilot/ardupilot #8419: Why choose this algorithm to calculate break angle?

aogrcs, 2018-05-17:
#### Issue details
In the function void Copter::ModePosHold::poshold_update_brake_angle_from_velocity(int16_t &brake_angle, float velocity), there is an algorithm to calculate lean_angle:
lean_angle = -poshold.brake_gain * velocity * (1.0f+500.0f/(velocity+60.0f));
Why choose this? Is there any reason or experience? Thanks!
Sincerely

#### Version
master

#### Platform
Copter

#### Airframe type
quadrotor

#### Hardware type
pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2020-12-02:
Hope you found your answer for this!

Please direct questions like this to our Discord or Discuss venues as a first step.

closed 2020-12-02 by peterbarker

rmackay9, 2020-12-02:
I must say that I do not know why we use this particular formula.  I think the original developer (who isn't around much anymore) came up with it after much trial and error but I could be wrong.  Perhaps there was a formula in mind when he created it.