ArduPilot/ardupilot #8437: Ublox GPS jamming indicator

geofrancis, 2018-05-19:
Would it be possible for ardupilot to read the Ublox GPS jamming indicator and throw a warning if it gets high? I have recently had issues with a gps being jammed by a companion computers camera cable. but everything on ardupilot looked fine with a decent 3d fix but it couldnt get RTK fix due to the noise.

WickedShell, 2018-05-21:
So I've considered this in the past. Unfortunately the jamming indicator is only useful when you have a known baseline, and this baseline will be different for every aircraft. If you build 20 of the same thing it works great, otherwise you would need to find the appropriate value for each vehicle individually (in a minimum noise/jamming environment). Because I can't see anyone doing this, I haven't considered it a worthwhile feature to add in the past.

magicrub, 2018-05-21:
@WickedShell but you could "calibrate" this letting it sit outside, or do a test flight, in known non-jamming conditions and then look at logs to determine a threshold to set a param with

WickedShell, 2018-05-21:
@magicrub it has to be a test flight (you need to actually power/spin all your components). I'm just not sold that it's worth the effort because very very few people will ever bother to set it up.

magicrub, 2018-05-21:
I know one person who probably would (cough) @geofrancis (cough)

geofrancis, 2018-05-21:
What about warning if it hits 100%? Or just logging it?

WickedShell, 2018-05-21:
Do you have a log where it maxes out? (I've never seen a log where it actually capped)

geofrancis, 2018-05-21:
I will need to dig one out,or recreate it. But if any of you have a pi camera v1 you can see yourself how hard it hits the GPS.

WickedShell, 2018-05-21:
I have a number of things here that are very hard on the GPS, but not one of them has ever managed to usefully spike the indicator to max (which is why I'm questioning it)

suraqis, 2018-05-21:
I saw few logs from GPS jammed flights and I think it's not a good idea to rely on u-blox's jamming detection features. You may be not only jammed but also spoofed. The main indicator of GNSS external interference is its inconsistency with inertial measurements and high EKF positioning variance. Also, two GNSS receiver configurations are tricky because jamming may have different success on both receivers and EKF will try to "average" them. This may lead to navigation problems and in case of mag autolearn or airspeed autocalibrate may lead to flight control problems.

geofrancis, 2018-05-22:
https://jeanleflambeur.wordpress.com/2016/07/29/raspberry-pi-camera-v2-gps/

geofrancis, 2018-05-22:
I'm currently in hospital 14" prop made it half way through my finger bone lol so I can't test atm

blazczak, 2020-11-06:
Found this thread while troubleshooting a similar issue. How did you get it fixed @geofrancis (the companion computer camera cable interference, not the finger - although both sound intriguing heh). Was it a CSI/MIPI FFC? I'm going to try to shield my flex cable using a household method unless someone who's run into this before has a better suggestion.

closed 2022-01-18 by geofrancis

rmackay9, 2022-09-01:
@mrx23dot,

We normally judge the GPS quality using the GPS provided accuracy numbers in particular the speed accuracy is one of the earliest indicators that something is going wrong.  These numbers can be viewed in real-time I think but in any case they certainly show up in the onboard log's GPA message.  Speed Accuracy is GPA.SAcc.  There's also height accuracy (HACC) and Vertical position accuracy (VACC).