ArduPilot/ardupilot #8448: Rover: accel/decel limiting issues

rmackay9, 2018-05-21:
There are at least four cases where the acceleration limiting is not working properly:

1. it immediately decelerates to zero after reaching a waypoint
2. it does not correctly decelerate before reaching a turn on pivot-steering vehicles
3. while in Auto (but disarmed) the desired_speed is allowed to accelerate up to full speed meaning the vehicle immediately jumps to full speed once armed
4. RTL-within-Auto jumps to Hold mode as soon as the vehicle reaches the wp-radius around home which leads to throttle jumping immediately to zero

rmackay9, 2018-05-21:
1 and 2 are resolved in this branch: https://github.com/rmackay9/rmackay9-ardupilot/commits/rover-stop-fix

IamPete1, 2021-08-30:
@rmackay9 any update? Was that branch mereged? it seems to be gone.

`calc_reduced_speed_for_turn_or_distance` function is not in master anymore.

rmackay9, 2021-08-31:
@IamPete1, thanks for finding this.  I've just re-tested and as expected the top two issues have also been resolved so I'll close this.

Below is a picture dhowing the desired speed gently climbs and descends.
[image]

closed 2021-08-31 by rmackay9