ArduPilot/ardupilot #8456: Intel AeroRTF can not access Nuttx system console

trgiman, 2018-05-22:
#### Issue details
Can not access system console - the Nuttx console from serial port. FTDI cable connection

#### Version

3.5.5

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Intel Aero RTF

#### Hardware type
Intel Aero RTF AscTech - Pixhawk architecture

#### Logs

1.I did switched off Mavlink running on UARTC_DEFAULT_DEVICE "/dev/ttyS3" // Telem using SERIAL1_PROTOCOL PARAM. I did set correct baud rate trough SERIAL1_BAUD param.

2.No Mavlink is running on UARTC_DEFAULT_DEVICE "/dev/ttyS3"

3. In nuttx-configs/aerofc-v1/nsh/defconfigs set CONFIG_UART5_SERIAL_CONSOLE=y and CONFIG_UART6_SERIAL_CONSOLE is commented

But it seems nuttx is is not forwarding nsh system console to UARTC_DEFAULT_DEVICE "/dev/ttyS3"

peterbarker, 2020-12-02:
Nothing can access nsh now-adays as we're no longer using NuttX!

Thanks for the report, 'though.

closed 2020-12-02 by peterbarker

lucasdemarchi, 2020-12-02:
And aero etc isn't supported anymore...