ArduPilot/ardupilot #8472: Rover: add drone follow mode

rmackay9, 2018-05-23:
We have implemented follow on Copter, we should also implement it on Rover.

The first couple of steps will be:

1. integrate the AP_Follow library into Rover (i.e. add it to the build, declare it, call it's init function)
2. create a new [Follow mode](http://ardupilot.org/dev/docs/rover-adding-a-new-drive-mode.html).
  - similar to Copter, the new Rover follow mode should take the other vehicle's velocity vector and turn that into a heading and speed request.  Guided mode already implements this heading-and-speed control so it can be used as a reference for the new mode.

JimiJoplin, 2018-07-09:
Adding new drive mode: STEP 2
class ModeGuided is the reference for the new class ModeFollow.

https://github.com/ArduPilot/ardupilot/pull/8858/commits/d1b3ed3e60d356cdff532266c54682aa8b7e44b5

rmackay9, 2018-07-09:
@JimiJoplin, just FYI - we already have a [PR with follow mode in this PR](https://github.com/ArduPilot/ardupilot/pull/8620).

rmackay9, 2018-08-22:
This is in master now so closing.

closed 2018-08-22 by rmackay9