ArduPilot/ardupilot #8474: Vehicle keeps climbing unexpectedly

expntly, 2018-05-23:
#### Issue details
Our vehicle hovers at 40% in QSTABILIZE.  When switching to QLOITER, it starts climbing.  We've read http://ardupilot.org/copter/docs/ac_throttlemid.html and adjusted our Q_M_THST_HOVER to 0.4, and tried Q_M_HOVER_LEARN=2, but that doesn't help.  We've disconnected the lidar to make sure this wasn't the problem.  Both barometers seem to work correctly.  QTUN.Alt and QTUN.DAlt seem suspicious.

From the documentation, it is unclear how to command a vehicle to hover in QLOITER (LEARN enabled or disabled).  It was more clear before when there was a midpoint and a deadzone.

#### Version
AP-3.8.5beta1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
X8

#### Hardware type
PH2.1

#### Logs
DF log: https://drive.google.com/open?id=1RjurSjVenTJ4g7qupCl4kVuZyjZ7NYBI

kd0aij, 2018-05-23:
hover throttle learning is not implemented for quadplanes, and Q_M_THST_HOVER is used only in initializing the integrator in the Z controller (on transitions)
Also, I don't think the copter documentation on mid-throttle adjustment is relevant to quadplane.

If you are hovering at 40% throttle stick, I'd expect it to descend on transition to QLOITER, though. Does the climb persist, or does the rate eventually match the position of the throttle stick?

expntly, 2018-05-23:
> hover throttle learning is not implemented for quadplanes

What's the use of [Q_M_HOVER_LEARN](http://ardupilot.org/plane/docs/parameters.html#q-m-hover-learn-hover-value-learning) then?  This is confusing now.

When you say "on transition", do you mean fixed-wing to VTOL and back?

The climb persists.

kd0aij, 2018-05-23:
Sorry, the parameter exists because quadplane uses the multicopter controller, but is ignored because the hover learning code isn't called by quadplane.  

I haven't flown any quadplanes myself, just tailsitters, but I have implemented hover throttle learning in my branch for quad tailsitters:  https://github.com/ArduPilot/ardupilot/pull/8113
It seems to work correctly for tailsitters, but is disabled (and untested) for all other frametypes.

But if your quadplane is climbing with the throttle stick centered in QLOITER mode, there must be something else wrong. Hopefully the log you posted will provide some clues...

expntly, 2018-05-24:
> with the throttle stick centered

So there is a mid point and deadzone?  Are they RC3_* params?

expntly, 2018-05-24:
Is Q_M_THST_HOVER also ignored?  Looking at the code, it looks like it depends on the LEARN param being enabled anyway.

https://github.com/ArduPilot/ardupilot/blob/88dc17fe6e8fa9909e0da6799d03132201d00c47/libraries/AP_Motors/AP_MotorsMulticopter.cpp#L421

It would be great to have this documented, we spent hours trying to adjust these parameters.

expntly, 2018-05-24:
I've sent [this PR](https://github.com/ArduPilot/ardupilot_wiki/pull/1257) to fix Q_THR_MID documentation.

I'd be glad to also make changes to http://ardupilot.org/plane/docs/parameters.html#q-parameters to document `Q_M_THST_HOVER` and `Q_M_HOVER_LEARN` being ignored, but I'm not sure where to find it in the repo?  The "edit on github" link is broken.  Also let me know if there are other parameters in the same situation.

kd0aij, 2018-05-24:
I'm not aware of any parameters for setting throttle curves in the quadplane code.

Q_M_THST_HOVER is used in initializing the integrator in the Z controller (on transitions)
That's in method run_z_controller() of quaplane.cpp

I guess you have to fork the ardupilot_wiki repo to edit it; at least that's how I've done it

WickedShell, 2018-05-24:
All parameter documentation comes directly from the code, you'd have to actually make an edit [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsMulticopter.cpp#L130) note that this applies to both quadplaneplane and copter

expntly, 2018-05-25:
PR: https://github.com/ArduPilot/ardupilot/pull/8490

So for Q_M_THST_HOVER, what does the pilot need to do to hover?  Say if thrust to hover is 25%.  I'm not seeing any deadzone param, and trying to keep exactly 25% from the throttle stick seems challenging.

auturgy, 2018-06-07:
We might be jumping to a solution here without stepping back and getting some context on the problem.  Is it worth considering whether hover throttle learning might be useful in quadplane?  @tridge

kd0aij, 2019-01-05:
@auturgy @Naterater  For hover throttle learning in quadplanes, see https://github.com/ArduPilot/ardupilot/pull/10028
I find it useful for tailsitters.

@Naterater As I said above I'm not aware of any support for throttle curves in Plane. If you can find it, please let us know :)

IamPete1, 2021-02-08:
We now have hover learning on plane so I'm going to close this. https://github.com/ArduPilot/ardupilot/pull/10323

closed 2021-02-09 by IamPete1