ArduPilot/ardupilot #8479: AP_RangeFinder: TerraRangeI2C should use non-block semaphore in update()

rmackay9, 2018-05-24:
As discussed in [this PR related to the Maxbotix I2C rangefinder driver](https://github.com/ArduPilot/ardupilot/pull/8471), the TeraRangerI2C driver should take use semaphore using non-blocking instead of BLOCK_FOREVER from within the [update function](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_TeraRangerI2C.cpp#L156).

The reason is that the update() function is called from the mainloop so in the unlikely case that the I/O thread locks up, this would cause the main thread to also lockup.  Of course, the I/O thread locking up would likely lead to a crash anyway as I think we would not receive updates from the main IMU either.

kunalkk308, 2018-10-12:
I would like to contribute to this, is this still active?

divyaS09, 2023-03-20:
I would like to contribute to resolving this issue. Please assign this issue to me.

closed 2026-05-13 by rmackay9