ArduPilot/ardupilot #8483: waf build error due to missing quote character in auto-generated path

SteveJos, 2018-05-24:
#### Issue details
Installed GNU Tools ARM Embedded 7 2017-q4-major toolchain in the default path (contains white spaces per default).
Configured with ./waf configure --board=fmuv3
It correctly detected the compiler with no errors or warnings.
After that ./waf copter compiled all files until the very last one.
While generating the apj file it raised the following error:

#### Version
Master

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

`Checking for env.py
env added CHIBIOS_BUILD_FLAGS=USE_FATFS=yes CHIBIOS_PLATFORM_MK=os/hal/ports/STM32/STM32F4xx/platform.mk CHIBIOS_STARTUP_MK=os/common/startup/ARMCMx/compilers/GCC/mk/startup_stm32f4xx.mk
env added HAL_WITH_UAVCAN=1
[502/503] Linking build/fmuv3/bin/arducopter
[503/503] Generating bin/arducopter.apj
/bin/sh: -c: Zeile 0: Syntaxfehler beim unerwarteten Wort `('
/bin/sh: -c: Zeile 0: ` /cygdrive/c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-objcopy -O binary bin/arducopter bin/arducopter.bin &&     python '/cygdrive/d/Projekte/Multicopter/ardupilot/Tools/ardupilotwaf/px_mkfw.py' --image 'bin/arducopter.bin' --prototype '/cygdrive/d/Projekte/Multicopter/ardupilot/build/fmuv3/apj.prototype' > 'bin/arducopter.apj' '

Waf: Leaving directory `/cygdrive/d/Projekte/Multicopter/ardupilot/build/fmuv3'
Traceback (most recent call last):
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Scripting.py", line 266, in run_commands
    ctx = run_command(cmd_name)
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Scripting.py", line 250, in run_command
    ctx.execute()
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Scripting.py", line 616, in execute
    return execute_method(self)
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Build.py", line 255, in execute
    self.execute_build()
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Build.py", line 275, in execute_build
    self.compile()
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Build.py", line 377, in compile
    raise Errors.BuildError(self.producer.error)
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Errors.py", line 45, in __init__
    WafError.__init__(self, self.format_error())
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Errors.py", line 51, in format_error
    txt = tsk.format_error()
  File "/cygdrive/d/Projekte/Multicopter/ardupilot/modules/waf/waflib/Task.py", line 415, in format_error
    txt = cmd
NameError: global name 'cmd' is not defined
`

jregenstein, 2018-05-31:
I had a similar error while trying to compile in cygwin. It turned out I was missing a dependency, installing the python future module fixed it.

BrightSoulXYHY, 2019-05-18:
same error in Cygwin of Building Copter-3.6.8 for /waf configure --board=fmuv3

IrisGreen, 2019-07-04:
same error in Cygwin of building copter-3.6 for ./waf copter:
`global name 'cmd' is not defined`
already installed python future but didn't work

galdalali, 2019-10-26:
Did someone manage to solve this issue?

juri117, 2019-11-06:
For me this error was raised after I changed the branch.
I forgot the "--recursive" when updating the submodules, this fixed it:
`git submodule update --recursive`

BrightSoulXYHY, 2019-11-06:
sometime it dosen't work for net work error
I solve it also by `git submodule update --recursive`
If it doesn't work del all dir in module/ and try again
thank for helping

IamPete1, 2021-07-26:
Sounds like this was resolved by updating sub-modules.

closed 2021-07-26 by IamPete1