ArduPilot/ardupilot #8484: Frsky Passthru library does not report vertical speed when in DCM mode

Hwurzburg, 2018-05-24:
All versions:
When in DCM mode, FRSky passthru does not report vertical speed (climb rate).
At least baro rate should be reported in this mode.
It is reported in VFR_HUD mavlink packets.

Discovered flying F4light HAL version on Omnibus F4 Pro since it must be flown in DCM mode until EKF issues with baro glitches are resolved.

floaledm, 2018-05-24:
Hi Henry,

Thanks for reporting this!

It might be a good idea to first discuss whether DCM mode should be supported here. In essence, maybe we should think twice about a fix that will be resolved by a larger fix?.. as we may end up introducing code that ultimately will not be useful (once the larger fix is issued). I think it would be good to get some more perspective before anyone works on a PR to address this. Just my thoughts...

Hwurzburg, 2018-05-24:
DCM has always been a fallback mode for EKF operation.....why stop sending telemetry in a valid mode?

Hwurzburg, 2018-05-24:
not sure of order of fallback, but whatever excessive innovations/divergence contributed to the EKF to stop aiding (excessive vibs, GPS dropouts, baro shift,etc.), its always possible to fallback to DCM temporarily in flight....thats why its still there and hasnt been removed...at least that is my understanding....

WickedShell, 2019-06-27:
Looked into this briefly, as long as GPS is available then DCM will report a velocity NED which is how FrSky is getting the vertical speed, however if you are in DCM because you don't have GPS, then DCM does not have an estimate to provide you.

yaapu, 2019-06-27:
why not fall back to baro climb rate like in the OSD code?
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_OSD/AP_OSD_Screen.cpp#L726

```
    if (ahrs.get_velocity_NED(v)) {
        vspd = -v.z;
    } else {
        auto &baro = AP::baro();
        WITH_SEMAPHORE(baro.get_semaphore());
        vspd = baro.get_climb_rate();
    }

```

closed 2020-02-20 by tridge