ArduPilot/ardupilot #8489: Rover: add DPTH message to record echosounder depth and lat/lon

rmackay9, 2018-05-24:
We have [added support for echosounders](http://ardupilot.org/rover/docs/common-echologger-ect400.html) (aka underwater sonar).  To make it easy to import the data to mapping tools like [ReefMaster](https://reefmaster.com.au/) we should log the vehicle's latitude, longitude and echosounder depth to a new message called DPTH (or similar).

Some advice to help with implementation:

- [wiki page on adding new dataflash log message](http://ardupilot.org/dev/docs/code-overview-adding-a-new-log-message.html) (use the easy way)
- add a new function called Log_Write_Depth() to Rover's Log.cpp
- get lat and lon from current_loc.  we need to be careful not to write these to the logs as floats because if we use regular floats (as opposited to doubles) we will lose precision.  So we should use doubles or longs (i.e. 32bit integers).  probably the latter is more consistent with how we log lat/lon in other messages.
- only log the new method if the vehicle is a boat and the range finder is pointing downwards.  We don't want to be logging a forward facing lidar accidentally.
- first field of a logging message is always the time.

By the way ReefMaster allows importing from a CSV file which is the method I think we will use.  There will probably need to be some post processing on the dataflash log to extract these new messages although ardupilot's log download also supports only downloading a single message from a log so perhaps we will use that.

Some might argue that we already record the vehicle's position and the rangefinder's distance in other messages but the data is not in the same message meaning there's a slight timing distance.  I'm also trying to make it as easy as possible for the user so I think this dedicated message will be helpful.

rmackay9, 2018-05-25:
@WickedShell, I think we can avoid the majority of the post-processing if we have this message.  So the worst case procedure could be done with only the mission planner and excel:

- download the log
- convert from .bin to .csv
- open in excel and filter for the new DPTH messages
- resave as .csv

So this is something that everyone can do without any extra tools.  Without the message we definitely need to write a tool to do the interpolation between position messages and rangefinder messages.

WickedShell, 2018-05-25:
@rmackay9 Right, but it add's more parameters (as now the user needs to note per rangefinder if it's a depth sensor or not), which adds a fair amount of extra setup/complexity. Otherwise you will just be logging multiple entries of rangefinder data. It will also require angle correction being done on the vehicle before logging.

rmackay9, 2018-05-25:
@WickedShell .. add a note just as you replied.. no need for new parameters and no correction for now at least.

rmackay9, 2018-06-13:
This is in master and being beta tested as part of Rover-3.4.1-rc1.

closed 2018-06-13 by rmackay9