ArduPilot/ardupilot #8503: Forward to waypoint # when failsafe triggered

joshzhangca, 2018-05-26:
#### Issue details
_Please describe the problem, or desired feature_
Current only have RTL, hold etc., it is great if it can be set a waypoint# when failsafe triggered
#### Version
_What version was the issue encountered with_
Rover3.3
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[x  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk 2.1
#### Logs
_Please provide a link to any relevant logs that show the issue_

khancyr, 2018-05-26:
I don't remember the state of rally point in rover, but it should definitvly be an anwser for this !

rmackay9, 2018-05-27:
Txs for the enhancement request.  It's a good idea.  Our Rally point code definitely doesn't handle this at the moment.  It is possible that the advanced failsafe code does but it's not enabled by default.

Anyway, a good idea.

IamPete1, 2021-09-07:
This is the `DO_LAND_START` functionality that we now have on plane and copter.