ArduPilot/ardupilot #8506: Rover : Scheduled Mission Plan

Lumono, 2018-05-26:
#### Issue details
_Please describe the problem, or desired feature_
It is very useful if the Mission Plan can be scheduled. E.g. the plan can start at May 30, 2018.
It should be like the scheduled task in Windows. A schedule can be recurring.
#### Version
_What version was the issue encountered with_
Any versions 
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  x] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2018-05-27:
Yes, great idea and it is fairly easy to add.  All this requires is supporting the NAV_DELAY mission command that we already support in Copter.  I can't promise when this will be done but I'm sure it'll go into a future release.

Lumono, 2018-05-27:
It is easily if the Mission Plan is a object, then it can be scheduled, can be loaded by file from disk, so Rover can run multiple mission plan, also support Python load.

magicrub, 2019-08-14:
implemented via https://github.com/ArduPilot/ardupilot/pull/12038

closed 2019-08-14 by magicrub