ArduPilot/ardupilot #8515: throttle goes to full when disarmed quadplane

igal74, 2018-05-28:
#### Issue details
when going from arm to disarm throttle goes to full

#### Version
3.8.5 release

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ v] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
quadplane

#### Hardware type
pixhwak2 (cube)

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2018-05-28:
Log please

igal74, 2018-05-29:
[pixhawk2log.zip](https://github.com/ArduPilot/ardupilot/files/2047277/pixhawk2log.zip)
good morning,
first, thank you for the quick response!!
2nd, this morning it seems to be open no matter what i do - arm / disarm - it stays on pwm output 1919 (maximum)
i've attached a zip of the log

magicrub, 2018-05-30:
Sounds like a configuration error. Please post on discuss.ArduPilot.Org.

closed 2019-01-05 by Naterater