ArduPilot/ardupilot #8517: Support Follow Me using Copter 3.6 Swarm follow code as basis

DonLakeFlyer, 2018-05-28:
I'm asking because from the QGC side of things the current code for Follow Me which only works on PX4 seems close to be able to supporting this on ArduPilot as well.

The way QGC currently works:
* It sends a FOLLOW_TARGET message to the vehicle when the vehicle is a Follow flight mode.

Possible way to make it work:
* FOLLOW_SYSID == 0 or -1 indicates that the system is in Follow Me mode, not swarm mode.
* The default for FOLLOW_SYSID is -1 to indicate Follow Me support since I think that is more common end user usage.
* ArduPilot supports FOLLOW_TARGET when in this mode and in Follow flight mode
* QGC sends FOLLOW_TARGET when in Follow flight mode and FOLLOW_SYSID=-1
* ArduPilot uses it's FOLLOW_* parameters with respect to adjusting the vehicle to the position in the FOLLOW_TARGET message

OXINARF, 2018-05-28:
This was discussed in Gitter, starting at about https://gitter.im/ArduPilot/ardupilot?at=5b0c39b64eaffb692d611b3a

My opinion about it is that QGC should support the MAV_CMD_DO_FOLLOW command and then:

> we change the parameters to whatever is sent from DO_FOLLOW command and then automatically enter the Follow mode. If we enter the follow mode first then it should follow whatever is in the parameter

DonLakeFlyer, 2018-05-28:
Copying from Gitter:

```I know nothing about those other commands. Nor do I know anything about the discussion that happened around them. I did not implement follow me for px4 in QGC it was a contribution from someone else.```

@DonLakeFlyer That just reinforces my point about not making sense that whoever implemented it didn't use the proper commands. If you want to make things more portable between vehicles, using the existing command is what makes sense, not having to know the mode number that is Follow.

I don't disagree with that. I'm just trying to piggyback on an existing feature such that it requires little change. If someone is interested in working through a complete follow me protocol I'm happy to work through that. Keeping in mind that will likely blow past Copter 3.6 timeline in an instant.

```Your missing my point which is that my opinion is Follow Me is a more common end user capability that will be used than swarm follow me. Hence by default you want the firmware setup to be in that mode. You also don't want to force the user to know things like ground station ids which they know nothing about to use follow me. I"m trying to improve the ease of use of Follow Me.```

@DonLakeFlyer It might be more common but we can't just dismiss the other use cases. See an example: an user starts using ArduPilot with QGC, changes to Follow mode and it automatically follows him; he decides to try swarm, everything works great; now he wants to have the vehicle follow his device using QGC and it doesn't work. How does the user know what is wrong? We would have to document that with QGC he needs to set SYSID to -1 or it doesn't work - that's going to be missed by many and brings GCS-specific behavior without any need.

That would be covered in the Follow setup page in QGC which specifies whether you are using follow for Follow Me or Swarm.

```Why can't QGC be changed to use the proper MAVLink protocol?```

Because who really knows what that is? The mavlink protocol is piled high with things that nobody has ever supported. The proper mavlink protocol tends to be what gets implemented. So my guess that whatever the follow me commands that are in mavlink now are wrong since nobody has every implemented them.

In general I agree that it would be better to do this the right way. That just tends to be way harder and longer. If there is interest in doing that then lets do it.

rmackay9, 2018-05-28:
I think it would be good to support some of the mavlink follow messages.  It's written without that support but the intention was to add support eventually.

peterbarker, 2018-05-29:
On Mon, 28 May 2018, Don Gagne wrote:

>  *  It sends a FOLLOW_TARGET message to the vehicle when the vehicle is a
>     Follow flight mode.

We use GLOBAL_POSITION_INT as it means we can follow any vehicle without 
having that vehicle need to understand it is being followed.  Bit creepy, 
but there you go.

>  *  QGC sends FOLLOW_TARGET when in Follow flight mode and FOLLOW_SYSID=-1
>  *  ArduPilot uses it's FOLLOW_* parameters with respect to adjusting the
>     vehicle to the position in the FOLLOW_TARGET message

What do you do when these parameters aren't available - does PX4 
Firmware have the equivalent?

What fields in FOLLOW_TARGET do you expect the autopilot to respond to?

The FOLLOW_TARGET message doesn't have a target systemid/componentid. 
That seems a little... odd.... does this end up as "everyone that can see 
this go here" message, then?

How is FOLLOW_TARGET any better than guided-position or guided-velocity - 
which I'm guessing you *already* support for follow-me?

The message is also odd in a few other ways:
  - the bits specifying supported data look a little odd
  - the unspecified covariances
  - the buttons things?!
  - 44 bytes!

Kelly-Foster, 2018-06-03:
Hopefully us Plane folks won't be left behind on this!

Kelly

rmackay9, 2019-09-30:
I think we could potentially close this now because it works with Rover and so it should also work with Copter although it should be tested perhaps as part of the upcoming 4.0.0 beta.

IamPete1, 2021-09-12:
@rmackay9 sounds like this can be closed?

IamPete1, 2022-03-03:
We now support `FOLLOW_TARGET` via https://github.com/ArduPilot/ardupilot/pull/20177

closed 2022-03-03 by IamPete1