ArduPilot/ardupilot #8525: Create ardupilot-gcs@googlegroups.com

peterbarker, 2018-05-29:
#### Issue details

When we make changes to the mavlink stream we send (e.g. https://github.com/ArduPilot/ardupilot/pull/8522), it would be great to be able to send a message to a list received by the authors of the various GCSs that support ArduPilot.

That way when we break something we can point out that we *did* say.  And apologise.

I propose a ardupilot-gcs@googlegroups.com mailing list which GCS authors can subscribe to on the understanding it's a low-volume list that might give them a heads-up on changes (and possibly a chance to argue the changes down).  It would not be used to request changes to AP - we have other mechanisms for that.

auturgy, 2018-05-29:
+1
Good idea

peterbarker, 2018-05-30:
```
Don Gagne @DonLakeFlyer 12:40
Something like that would be awesome. I know it's annoying to work with me because I never attend anyones dev calls or anything like that. So something like that would work great for me.
Since I tend to miss lots of stuff since I'm not looped in unless someone pings me.
```

peterbarker, 2018-06-05:
https://groups.google.com/forum/#!forum/ardupilot-gcs

@auturgy Could you poke the GCS authors you tagged the other day, please?

closed 2018-06-05 by peterbarker