ArduPilot/ardupilot #8526: Rover: Support heading from two RTK GNSS

mikerob, 2018-05-29:
#### Issue details
RTK GNSS are sufficently accurate that you ought to be able to get a superior heading from two units compared to using the magnetometer - which can be particularly problematic on Rovers. 

It would require the ability to set the location of both GNSS units, and (probably) an accuracy threshold (error uncertainty calculated from horizontal accuracy and baseline length).

This could also be applicable to Plane.

#### Version
All

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[ X ] Rover
[  ] Submarine

#### Airframe type
Rover, aircraft

#### Hardware type
Cube / Here+

#### Logs
N/A

proficnc, 2018-05-29:
And copter, and boat.

geofrancis, 2018-05-29:
https://github.com/ArduPilot/ardupilot/issues/3934