ArduPilot/ardupilot #8538: Splinewoypoint: skips ahead on mode change

AndKe, 2018-05-30:
#### Issue details
When next wp is a splineWP, it will be skipped on auto resume.
This is potentially dangerous if there is an obstruction in the "shortcut" path.

#### Version
AC3.6.RC1 and older AC

This video illustrate is perfectly:
flying a box, when AC mode is temporarily changed, then back to AUTO, the current WP will be skipped. 

https://www.youtube.com/watch?v=v6GIQwG1rlE

rmackay9, 2018-05-31:
well, that's unexpected!  thanks for the report!

AndKe, 2018-05-31:
I tried to solve it and push a PR at the same time, but had to give up as I failed to find the reason within reasonable time. :)

khancyr, 2018-05-31:
@rmackay9, I suspect this is the behavior caused by : https://github.com/ArduPilot/ardupilot/issues/6798

rmackay9, 2018-05-31:
@khancyr, OK thanks.  If that's the case then it should be fixed in master (and Copter-3.6)..

khancyr, 2018-05-31:
@rmackay9 I mean the patch skip the first spline waypoint is case two waypoints are too close, 
 // if target velocity is zero the origin and destination must be the same
// so flag reached destination (and protect against divide by zero)
I may explain the problem here

IamPete1, 2021-08-30:
I cant reproduce on master, I suspect this was fixed with all the SCurve changes.

closed 2021-08-30 by IamPete1