ArduPilot/ardupilot #8543: Adding a "tandem Fly" feature.

Rusty105, 2018-05-31:
I am sort of at a loss where to put this "issue" it's really more of a feature request, but it will span many projects, MP, Arudpilot, Mavproxy, various OSD projects. 

What I want to do is add a "Tandem Fly" feature, if it is not already implemented, I looked and didn’t see anything. What the feature will add is the ability to display on the OSD, and hopefully ground stations like Mission Planner, the position and distance to other Autopilots with the feature enabled. There are other Autopilot systems that already do this, MyFlyDream does it, and I believe Vector has something in the works. On the OSD it displays: direction, distance, and altitude difference, on ground stations it would show a "friendly" aircraft, similar to how MP shows ADBS contacts now. While for me the feature has merit in the recreational FPV hobby, I believe it has commercial benefits as well.

The plan...
I think I have the hardware figured out.  Each aircraft would have a Raspberry Pi, or other single board computers, on the ground running Mavprovy to relay the positional data from one Raspberry Pi to another. In turn each Raspberry Pi will relay the data back to the OSD and GS of the other aircraft. The other systems typically only show one other "tandem", I am pretty sure this will be able to track multiple tandems, very similar to the way MP shows multiple ADSB aircraft on the screen.

Before I go further with my project, it would be good to know if anyone else is working on something like this, or if the feature exists somewhere buried in the code...

And, open to suggestions and ideas......

Rusty

#### Version
All

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Any

#### Hardware type
Hardware independent

Naterater, 2018-06-01:
This would be really neat in general to display targets on HUD's. Garmin professional aviation displays do this already, and it would be nice to have something similar both for the GCS and the OSD.

Rusty105, 2018-06-02:
Not really like one ground station monitoring multiple aircraft.

Picture two, or more, friends flying tandem in a large field. It's done all the time. The trick is to making sure you know where your friends are so you can find them, then follow once you see them in your video feed. MyFlyDream (another autopilot company) makes an antenna tracker that has a device that allows two pilots to share their positional data with each other. More than two can connect, but you only see one at a time. 
Anyway... What I am thinking is that Ardupilot can do the same, only better! Most FPV pilots don't use a Mission Planner type of ground station, so I would like to keep that as optional equipment. BUT, carrying around a RasPi, or similar, is no big deal. Telemetry data from 1 plane will relay down to a RasPi, and share that with other RasPis on the ground. Each of those will in turn relay the positional data to their respective aircraft. Once the aircraft has the positional data from the RasPi, it can display it on the OSD.

Has that helped?   is it similar to what can be done now?

Kelly-Foster, 2018-06-03:
I would prefer an air-to-air system, whereby both plans emit a mavlink stream of their current position, attitude, velocity, and current commanded maneuver (pitch, roll, yaw, throttle).  Receiving aircraft shows bug on OSD for location (or direction if off screen) of the other aircraft.  Goal is to get on his tail before he gets on yours.  Loss of style points for mid airs.  A lot of fun for Man-on-man, even more fun when men-on-men, meaning there are multiple aircraft in the air, and you either hand fly to the nearest would-be victim, or let the flight commander decide nearest reachable target, and pursue that aircraft.  Of course, toning down the air combat angle, following aircraft could simply approach and maintain a set distance in x,y,z from target aircraft.  Then you both flip a switch and Followed become Follower, and vice versa.

Kelly

Rusty105, 2018-06-03:
The air-to-air sounds interesting, but how? If using the Sik radios, don't they have to have the same net ID? the rest of the stuff you mentioned can be done with the ground RasPis. With the ground based system, each pilot can keep their Sik radio settings the way they have them. I am open to ideas...

Kelly-Foster, 2018-06-04:
How do you envision ground-based Pi's would talk to each other? What link and what "language"?  Mavlink over Wi-Fi, or Bluetooth? And how does pilot instruct his Pi?  Commands such as following on/off, incremental position tweaks, cloaking device on/off, target Death Star exhaust port, etc?  Or maybe you were only envisioning tracking/following by human pilot, and not autopilot system? 

Kelly

Rusty105, 2018-06-04:
As for the Pis communicating on the ground, I am bouncing a few ideas in my head. One of the limitations of the Pi is only having one Uart, BUT it does have 4 USB ports. One USB port will have a telemetry radio communicating with its aircraft. But the question is now how to communicate with the other Pis? One idea is have the Pis join an Ad-hoc network on the ground (using the built in WiFi on the Pi3). All the parameters will be pre-determined so that the Pis will hopefully "just connect". The other idea is to use an inexpensive Raspberry Pi Zero W, and have that act as an Access Point for all the other Pis. The protocol will be Mavlink for all communications. I have to check to see it Mavlink can be routed to multiple points, but i believe Mavproxy can do that.

I think you may have misunderstood how these are going to be used. The pilot will not have to instruct the Pi to do anything, and there are not any automated moves. The tracking is really just an aide for the human to find his fellow pilots. HOWEVER, that is not to say those features couldn't be added later. One particularly useful display may be a heads up showing when to release the torpedoes to hit the exhaust port, but in the end it was still a human who hit the target.

Rusty

Rusty105, 2018-06-25:
Ok, now I get the "air to air" option. Sometimes it takes a different view to understand how an option would be beneficial. I was reading an account of someone using the Pitlab OSD, which has a "Buddy Fly" option. And from what I can put together from a brief description, it looks like it uses radio modems, possibly similar to the 3DR style of telemetry radios, to basically broadcast positional data to other "Buddy Fly" radios. The story I read had the two users flying from different fields, a few KM away, and then meeting up in midflight. To do this on Ardupilot, it would be one more radio on board, unless there is an option for the telemetry radios to broadcast a message for all to hear, not just radios on the same network ID. Might have to look into that as well....