ArduPilot/ardupilot #8545: Implement country specific regulations

Dronotique, 2018-05-31:
#### Issue details
Hello everybody,

New french laws will appear in the beginning of July.
1 of this law could be implemented in Arudpilot or in a new hardware (Arduino) solution.
We will need to had on our UAVs a led blink with U morse code.

But I think the problem is more global. Because each country could have some specificities, how could we implement them. 
At the beginning, I thinked about created a fork. But some peoples told me that I'm wrong becaus eforks are not a right solution.

What do you think about?

#### Version
ALL

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
ALL
#### Hardware type
ALL

khancyr, 2018-05-31:
On the question of stupid regulation rules, I don't think it is to open source project to adapt themself, but to user to be in phase with their own rugulation.. 
Especially when it is about a non existing law... 
I don't think we should have a debat here as mavros has nothing to do with that... It is a bridge between mavlink and ROS, not an autopilot, so not concerned by those rugulations... 
I vote to close this topic before it get out of control on non-sense discussion

Dronotique, 2018-05-31:
The debat is here. 
In fact, there are 3 ways to show the problem.

- The master branch of the open source project implements specific laws
- A fork by country, French in this case, is created to implements country specificities. And their is 2 ways to manage this fork.
The fork can be managed by the open source, or by each user

In the second case, a lot of smalls UAVs manufacturers should not be able to.

auturgy, 2018-05-31:
You shouldn’t need to fork. If it was determined that country specific firmware is required, necessary changes could be added as compile time options. That would avoid the risk of fragmented development. I don’t think we’re at that point yet though.

As an additional consideration, the hardware can’t be locked down, so ultimately it would still be up to the user to have the correct firmware - ie no real difference between a bespoke build and a country specific parameter file.

Regards,

James

Kiwa21, 2018-06-01:
By the way this law is still in consultation and not yet voted. I agree not to put this in ardupilot firmware, users should take care of this themselves

khancyr, 2018-06-29:
French law didn't change as announced.... closing this topic until UE vote something

closed 2018-06-29 by khancyr

magicrub, 2018-06-29:
This won't be the last. I suggest we create an AP_Region library that allows you to enable features for local laws. Would be nice to select an enum for the country and then things get set that are registered specific.
Today it's lights, tomorrow could be VLOS or whatever

hiro2233, 2018-06-29:
Here, the regulations need a coffee cup too :v

hiro2233, 2018-06-29:
> AP_Region

NO.

proficnc, 2018-06-29:
AP_Region actually makes sense!