ArduPilot/ardupilot #8547: Copter 3.6.0-rc1 allows arming with motor interlock enabled

bnsgeyer, 2018-05-31:
#### Issue details
It has been discovered that Copter-3.6.0-rc1 allows arming of tradheli with motor interlock enabled.   This is considered to be a serious SAFETY issue for helicopters.  

#### Version
Copter-3.6.0-rc1

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
TradHeli

#### Hardware type
Demonstrated in Pixhawk and in SITL

peterbarker, 2018-06-01:
Can you outline the steps taken to reproduce in SITL, please?

parameter values set, interlock-channel-stat changes, points at which arming is attempted, that sort of thing.

rmackay9, 2018-06-11:
this is fixed in master and has been released with 3.6.0-rc2.

closed 2018-06-11 by rmackay9

BlackBlack777, 2020-11-28:
Hello! please tell me how to disable the function, motor interlock enabled. I plan to fly without a remote control through the QGroundControl application and without a remote control I cannot disarm it.

bnsgeyer, 2020-11-29:
@BlackBlack777 please check the ardupilot discuss website. I left you a private message regarding this.