ArduPilot/ardupilot #8560: Rover: simple path planning around obstacles

rmackay9, 2018-06-01:
Rover-3.4 supports two somewhat independent methods of object avoidance:

- "simple avoidance" attempts to stop the vehicle before hitting things ([blog](https://discuss.ardupilot.org/t/rover-3-3-object-avoidance-with-360-lidar))
- "dodge" method used in AVC competition is described [here on the wiki](http://ardupilot.org/rover/docs/rover-object-avoidance.html).

We should integrate these features together and/or update the "dodge" method to handle simple path planning while in autonomous modes (Guided, RTL, Auto, SmartRTL).   This could work something like this:

- the vehicle slows down (as it does now) so that it does not hit things
- if "dodge" is enabled the vehicle decides whether to turn left or right to avoid the object
  - this decision could be based on the lidar reading (i.e. a 360 lidar is being used, perhaps gaps can be seen)
- vehicle turns and follows the object/wall until a gap is found in the direction towards the target position.
- vehicle attempts to turn into the gap

This simple path planning will probably not be able to guarantee that the vehicle doesn't get stuck but it will be a start.

This avoidance logic should probably be handed off to a new high level controller similar to how a GCS can control the vehicle in guided mode.  This new controller should probably be in a new library called AR_PathPlanner or something similar.

khancyr, 2018-06-01:
it should be available on copter too as it is just moving in the same plan

rmackay9, 2018-06-12:
[image]
This is a diagram based on discussion with @tridge and @peterbarker which shows how we could enhance the proximity and avoidance libraries to provide some simple path planning and "memory" of obstacles.

rmackay9, 2018-06-12:
.. and here's another diagram showing how we could integrate with more powerful solutions running on the companion computer.  In this image I use "OpenKai" but it could be ROS or anything else.
[image]

rmackay9, 2019-09-17:
This is included in Rover-4.0.0 using BendyRuler and Dijkstras.

closed 2019-09-17 by rmackay9