ArduPilot/ardupilot #8568: Rover: do-reverse persists when mission is restarted.

rmackay9, 2018-06-03:
If the previous mission ended with the vehicle moving in reverse (because of a DO_SET_REVERSE mission command) then the next time the mission is run, it will begin in reverse.

To reproduce the issue:

- create a rover mission with a do-set-reverse (with 1 as the first field) command in it.
- run the mission
- remove to do-set-reverse from the mission
- run the mission again, it will run in reverse (which is incorrect).

The cause of the issue is the mode.h file's ModeAuto::_reversed parameter is not reset to false.  The simplest solution is probably just to add a call to set_reversed(false) in mode_auto.cpp's enter() method.  The only downside of this simple solution is that if the mission is restarted without leaving auto, which is possible using a mavlink command I think, the reverse setting will remain.

This issue was reported in [this Rover discussion](https://discuss.ardupilot.org/t/rover-3-4-0-rc1-released-for-beta-testing).

WickedShell, 2018-06-03:
You also might resume a mission while you were in a reversed segment, and automatically setting reversed to false is probably not good for that situation either, as now the vehicle is very much off the mission profile.

WickedShell, 2018-06-03:
Also worth noting this class of issues applies to all the vehicles for any of the do set speed commands and similar.

rmackay9, 2018-08-22:
This is resolved in master and will go out with Rover-3.5.

closed 2018-08-22 by rmackay9