ArduPilot/ardupilot #8573: Add BRD_SAFETYOPTION parameter

JIMWAR, 2018-06-04:
#### Issue details
PWM_SERVO_SET_SAFETY_OPTION_DISABLE_BUTTON_OFF flag can not be found in the project

#### Version
_What version was the issue encountered with_

#### Platform
Plane 3.8

hewu2008, 2018-06-05:
```
/** setup safety switch options. Bit 0 is disable of button for safety off->on. Bit 1 is for on->off */
#define PWM_SERVO_SET_SAFETY_OPTION_DISABLE_BUTTON_ON  (1U<<0)
#define PWM_SERVO_SET_SAFETY_OPTION_DISABLE_BUTTON_OFF (1U<<1)
#define PWM_SERVO_SET_SAFETY_OPTIONS		_PX4_IOC(_PWM_SERVO_BASE, 38)
```

JeyPi, 2018-06-07:
Arduplane version 3.8.5
Using pixhawk. The safety switch allows me to Arm, but it does not allow me to disarm.

In the wiki I found these settings for the BRD_SAFETYOPTION:
0 | ActiveForSafetyEnable
1 | ActiveForSafetyDisable
2 | Active When Armed

Nothink else. but after the update to the newest arduplane soft - V3.8.5 - the setting was set to 3, which is not mentioned by the wiky.
None of the above settings work for arming the plane using the switch. Only the "3" does. But hey I am still not able to disarm. Also not able to arm or disarm via the mission planner.

WickedShell, 2018-08-26:
It's a bitmask not list of values. Setting it to `7` allows you to use the switch to the safe state while armed. (It won't actually change the vehicle mode to disarmed though). This is a support question, if you have further questions please take it to discuss.ardupilot.org

closed 2018-08-26 by WickedShell