ArduPilot/ardupilot #8576: Could not enable safety switch when perform ESC calibration

askariz, 2018-06-04:
#### Issue details
Could not enable safety switch when perform ESC calibration and the LED on the safety switch remain fast blinking .

#### Version
current master

#### Platform
[  ] All
[  ] AntennaTracker
[ ✅ ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
copter hex

#### Hardware type
pixhawk

rmackay9, 2018-06-05:
@askariz, is this in Copter-3.5 or 3.6?  If it's 3.6 I think we've got a fix coming in -rc2.

askariz, 2018-06-06:
I think this problem no fix in Copter 3.6-rc2 I have tested following the 
### Manual ESC-by-ESC Calibration  
[from](url) http://ardupilot.org/copter/docs/esc-calibration.html
The safety can no be open when long press the button and LED still blink.

Quadsorus, 2018-06-07:
I was not able to do it in chibiOS or Nuttx  rc-2. I disabled the safety and still no luck.  I was unable find rc-1 after upgrading to rc-2 so I had to go back to 3.5 stable to finish auto calibration.

rmackay9, 2018-06-11:
yes, I've also confirmed this issue persists in -rc2.  So we will fix it!

rmackay9, 2018-06-11:
PR to fix the issue raised here: https://github.com/ArduPilot/ardupilot/pull/8623