ArduPilot/ardupilot #8582: QGC confused as to what to do when BRD_PWM_COUNT is not available?

DonLakeFlyer, 2018-06-06:
So QGC has a camera setup page while looks like this for Copter:
[image]

The dropdown on Output Channel is built from BRD_PWM_COUNT which indicates how many channels are available:
[image]

When you select a channel for Output QGC sets the SERVO_N_FUNCTION value appropriately.

The question is what happens when BRD_PWM_COUNT is not available? Does this means you can't set SERVO_N_FUNCTION?

DonLakeFlyer, 2018-06-06:
@rmackay9 @tridge Not sure what to do here?

rmackay9, 2018-06-06:
@DonLakeFlyer, it is still possible to set the SERVOx_FUNCTION where x = 15 or 16 even if the BRD_PWM_COUNT = 4.  the problem is just that the 15th/16th output can't actually be used for PWM outputs unless BRD_PWM_COUNT is increased to 6.  ..so it won't work but the user/gcs will still be able to set the parameter.

I guess it depends upon how easy we want to make it for the user to avoid hitting the issue.

The BRD_PWM_COUNT parameter is always available on the Pixhawk family of boards but not on all the other boards (linux, the new racer boards, etc).  Basically though I don't think trying to tie the BRW_PWM_COUNT to the SERVOx_FUNCTION setup is something we should try to do...

DonLakeFlyer, 2018-06-06:
I guess I'll pull out the Output Channel stuff from the ui then.

closed 2018-06-06 by DonLakeFlyer

DonLakeFlyer, 2018-06-06:
> I guess I'll pull out the Output Channel stuff from the ui then.

No, that's not quite right either. It just needs to be dumber.