ArduPilot/ardupilot #8592: SITL no longer compiles under macOS

charlie-mtz, 2018-06-07:
#### Issue details
SITL no longer compiles under macOS.
Commit f986f1366ff55859bb96764a09fc3e28d83ec776 adds SOCK_CLOEXEC option to the socket type. This is not standard unix and doesn't work on the Mac.

A possible solution to this would be to do it the old way.
Instead of:
```
fd = socket(AF_INET, SOCK_STREAM | SOCK_CLOEXEC, 0);
```
do:
```
fd = socket(AF_INET, SOCK_STREAM, 0);
fcntl(fd, O_CLOEXEC);
```

#### Version
master at commit f986f1366ff55859bb96764a09fc3e28d83ec776
#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
SITL

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
```
[368/493] Compiling libraries/AP_HAL_SITL/UARTDriver.cpp
../../libraries/AP_HAL_SITL/UARTDriver.cpp: In member function 'void HALSITL::UARTDriver::_tcp_start_connection(uint16_t, bool)':
../../libraries/AP_HAL_SITL/UARTDriver.cpp:225:52: error: 'SOCK_CLOEXEC' was not declared in this scope
         _listen_fd = socket(AF_INET, SOCK_STREAM | SOCK_CLOEXEC, 0);
                                                    ^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
```

charlie-mtz, 2018-06-07:
Ok.

lvale, 2018-06-07:
http://man7.org/linux/man-pages/man2/socket.2.html

charlie-mtz, 2018-06-07:
Submitted pull request https://github.com/ArduPilot/ardupilot/pull/8593.

OXINARF, 2018-06-10:
Fixed via #8593.

closed 2018-06-10 by OXINARF

barracuda156, 2023-11-06:
Maybe this could work instead: https://fjordtek.com/git/Fincer/openntpd-portable/commit/e3106f50ac730b480f4a9dc05ee02e32d318bef6