ArduPilot/ardupilot #8595: Plane V3.8.5 would not disarm using the safety switch

JeyPi, 2018-06-07:
Arduplane version 3.8.5
Using pixhawk. The safety switch allows me to Arm, but it does not allow me to disarm.

In the wiki I found these settings for the BRD_SAFETYOPTION:
0 | ActiveForSafetyEnable
1 | ActiveForSafetyDisable
2 | Active When Armed

Nothink else. but after the update to the newest arduplane soft - V3.8.5 - the setting was set to 3, which is not mentioned by the wiky.
None of the above settings work for arming the plane using the switch. Only the "3" does. But hey I am still not able to disarm. Also not able to arm or disarm via the mission planner.

Naterater, 2018-06-07:
It is a bitmask.

3 = 1+2 (so it works on bit 0 and bit 1) that means that the switch can enable and disable the safety state, but it cannot make the plane "safe" again once the plane is armed. This was done to prevent planes falling out of the sky due to safety switches accidentally actuating.

closed 2018-06-07 by Naterater

JeyPi, 2018-06-08:
Well I understand that. But I believe there could be another approach to that problem. I work with Arm and disarm a lot when setting my platforms just to prevent any prop accident. 
Could the disarm procedure be available only when the plane is not flying? It knows it is flying right? Than it would never ever disarm during a flight (airspeed, baro, gps,....)
While I understand the reason it was taken away, I still miss using it. Until there is some decision to be taken about it (lets see what others will think) I am downgrading to Arduplane version 3.8.4 being aware the safety switch could cause the plane to disarm during a flight.  If only the safety switch was made more safe :-)

Naterater, 2018-06-08:
This was the discussion on it. If you want to be able to arm and disarm like before, just change the parameter to 6 (so you can change the state while armed).

The flying estimator has been deemed not reliable enough to be used for safety switch purposes.

scopix, 2018-06-09:
I have tried to use parameter 6 as you suggested and the safety switch will still not switch back to safe once the plane is armed.

This is a significant safety issue for anyone hand launching with two hands in auto mode. If I need to engage the safety switch after the motor spins up but before takeoff, I now have to reach down to the transmitter which is on the ground. Previously I could simply press the safety switch which is conveniently located where my hand hold the wing.

Can you please reconfirm that parameter 6 is intended to allow a change of switch state when armed? If so is there a bug in the script that is preventing this? Or can anyone suggest any other parameters?

scopix, 2018-06-09:
Problem solved.... parameter 7 gives the correct behavior.

JeyPi, 2018-06-09:
Somebody should mention the parameters in the wiky.

JeyPi, 2018-06-09:
Somebody should mention the parameters in the wiky.
I mean what each number does. 1-7. Some of us are not thinking in bits guys.
2^0 (index 0)=1, 2^1(index 1)=2, 2^2(index 2, square)=4
2^0+2^1+2^2=1+2+4=7
2^0+2^1=1+2=3
The switch can enable and disable the safety state, but it cannot make the plane "safe" again once the plane is armed. We want this so the switch will not be able to disarm the plane in flight (after it was armed)

2^0+2^1+2^2=1+2+4=7
We want this to use the switch to cycle between the Arm and Disarm state which is perfect for ground handling and setting thinks. Once it is ready, I personally would change that back to 3. That way my plane will newer fail because being disarmed in flight. 
Sure others will have to make their choices how to use the switch. It is good the choice can be made.