ArduPilot/ardupilot #8615: All: "log list" stops working if ARMING_REQUIRE is set to 0

gmorph, 2018-06-10:
Build and upload Rover from master.
Connect using mavproxy (mav10 NOT mav20)
If I set ARMING_REQUIRE to 1 then run "log list" I get the list of dataflash log files.
If I set ARMING_REQUIRE to 0 then run "log list" I get nothing.

I tested in Mission Planner and it had the same issue so I assume the issue is in the ArduPilot firmware and not the ground station.

rmackay9, 2018-06-10:
This is a bit tricky, We don't allow downloading logs while the vehicle is armed because (I think) it causes lots of problems in the main loop rate and causes the EKF to go wonky.

gmorph, 2018-06-11:
Its possible this is a design feature.  You may not want to request the list of dataflash logs nor download a dataflash log whilst a vehicle is armed and flying around as it uses too much telemetry bandwidth.

peterbarker, 2018-06-12:
On Mon, 11 Jun 2018, Grant Morphett wrote:

> Its possible this is a design feature. You may not want to request the list
> of dataflash logs nor download a dataflash log whilst a vehicle is armed and
> flying around as it uses too much telemetry bandwidth.

Yep, working as designed.  We don't allow log operations while the vehicle 
is armed.

closed 2018-06-12 by gmorph