ArduPilot/ardupilot #8617: V.Servo log is off by x100

Naterater, 2018-06-10:
#### Issue details
V.Servo is now logged as .05 v instead of 5.0 v.
(my logs have the servo rail at 8.4v, but it logs as 0.084v)

#### Version
Plane 3.8.5 beta 1 chibios

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
N/A

#### Hardware type
Cube autopilot, Castle power supply

#### Logs
https://www.dropbox.com/s/nreknqh244o06q9/00000066.BIN?dl=0

WickedShell, 2018-06-10:
@Naterater this is an issue with whatever tool you used to look at the log.

[image]

Naterater, 2018-06-13:
@WickedShell,

I realized that I provided a link to the wrong file. Could you double-check this log? Go ahead and close again if it's good.

https://www.dropbox.com/s/ikuqniqgqodr109/00000008.BIN?dl=0

WickedShell, 2018-06-14:
You're actually on a newer log tool then I am. @peterbarker got the units wrong on that one by accident and applied a 1e-2 scaler that your log tool was aware of... #8638 resolves it.

peterbarker, 2018-06-14:
@Naterater Which tool are you using?

closed 2018-06-14 by tridge

Naterater, 2018-06-14:
I am using mission planner's dataflash log viewer. Did you guys check GPS speed as well?

PR #8638 (2018-06-14): DataFlash: Fix power units
Power was being scaled by 1e-2 for some reason. Fixes #8617