ArduPilot/ardupilot #8622: Copter-3.6: RC8 minimum is greater than trim pre arm failures

rmackay9, 2018-06-11:
when upgrading vehicles from Copter-3.5 to 3.6 I'm getting pre-arm failure messages as shown below and it's not being resolved by repeating the RC calibration.  My transmitter only has 7 channels.

> PreArm: RC16 minimum is greater than trim
> PreArm: RC15 minimum is greater than trim
> PreArm: RC14 minimum is greater than trim
> PreArm: RC13 minimum is greater than trim
> PreArm: RC12 minimum is greater than trim
> PreArm: RC11 minimum is greater than trim
> PreArm: RC10 minimum is greater than trim
> PreArm: RC9 minimum is greater than trim
> PreArm: RC8 minimum is greater than trim

WickedShell, 2018-06-11:
This is a GCS issue (and the check actually catching an invalid configuration). The `RCx_TRIM` is being set, but `MIN`/`MAX` are not being set. The problem wouldn't occur if the GCS was appropriately setting both min/max as well, or choosing to not set the trim.

Naterater, 2018-06-11:
This seems to be an issue with updating. I did not run an RC calibration, yet still ran into the pre-arm failure.

Yes, the GCS should be catching it; however it is being caused by updates.

OXINARF, 2018-06-11:
This isn't caused by updates. Our defaults for RCx_MIN/TRIM/MAX in 3.5 are respectively 1100, 1500 and 1900. In 3.6/master these are exactly the same, we didn't change anything. The only thing new is that we now detect bad settings in a pre-arm check.

If this issue happens after an upgrade it just means that, **before the upgrade**, the user had bad settings (whether they were set manually or by a GCS).
We shouldn't change these settings automatically, so there isn't anything that we can do here other than telling the user there is something wrong.

closed 2018-06-11 by OXINARF

allen26022482, 2018-10-28:
To fix this you need to go into parameters and change RCTRIM to be the same or greater than RCMIN. I think the default value is 987 and RCMIN is 1000.

Henri-H, 2019-06-09:
I got this same issue with 3.6.9 (63263f) and MP1.3.66 after updating from FW about 01/2019 and MP same time. Radio calibration is done after first seeing this issue and issue didn't vanish.
[image]

Henri-H, 2019-06-09:
Here are all parameters.
[Parameters.zip](https://github.com/ArduPilot/ardupilot/files/3269735/Parameters.zip)

rmackay9, 2019-06-09:
@Henri-H, I'm afraid this was caused by an issue in the Mission Planner.  Newer versions of the MP may resolve the issue if an RC calibration is done but if not, I'm afraid you may need to go into the full parameter list and set the RCx_MIN, RCx_TRIM, RCx_MAX to be valid (i.e. min less than trim, trim less than max)

Henri-H, 2019-06-10:
Ok, I try setting up parameters manually. Mission Planner seems already to be latest 1.3.66 25.4.2019. Another interesting detail is that why RC is checked when only INS is selected for arming_check

Henri-H, 2019-06-11:
Now it works! I changed all parameters manually which appeared on the flight data screen. This Servo9 used for parachute was the last one.
[image]

ibrahimqazi, 2021-09-21:
@rmackay9 hello Randy, I am also facing this issue, I have already the latest version of mission planner. Do RCx_MIN, RCx_TRIM, RCx_MAX mean min, max and trim values for each channel, like throttle, yaw, pitch and roll? Do I need to set them manually?