ArduPilot/ardupilot #8632: Rover: MOT_PWM_TYPE should be a per channel setting

gmorph, 2018-06-12:
#### Issue details
At the moment setting MOT_PWM_TYPE will change ALL channels to be whatever type is selected.  However on Rovers you may have a throttle ESC that requires a changed MOT_PWM_TYPE but a steering  servo that doesn't.  This is why it should be a per channel setting.

#### Version
All

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

#### Airframe type
Standard.  Probably all really.

#### Hardware type
Pixhawk

#### Logs
N/A

rmackay9, 2018-06-13:
Maybe we could get away with just setting the "output mode" for channels which have the function set to  motor, k_motor1, k_motor2, k_motor3, k_throttleLeft, k_throttleRight or k_throttle.

For anyone interested in implementing this, the change would be made in the APMrover2/AP_MotorsUGV.cpp file's AP_MotorsUGV::setup_pwm_type method.  the hal.rcout->set_output_mode takes a mask which controls which channels are affected.  so the change would involve creating a mask that has a "1" in each bit corresponding to the channels that have their function set to one of those listed above.

Naterater, 2018-06-13:
If I'm not mistaken, this kind of already exists in plane using SERVO_BLH_OTYPE and SERVO_BLH_MASK. The MOT_PWM_TYPE only exists as Q_M_MOT_PWM for planes.

And if I'm thinking logically, the MOT_PWM_TYPE will only apply to channels that are assigned to motor functions, not servos.

NickNunno, 2018-06-16:
This would be an excellent feature!

magicrub, 2018-06-17:
Jaime will be checking my branch here: https://github.com/magicrub/ardupilot/tree/issue8632_motor_bitmask

magicrub, 2018-06-17:
and this PR https://github.com/ArduPilot/ardupilot/pull/8671

rmackay9, 2018-07-13:
Resolved in master, thanks!

closed 2018-07-13 by rmackay9