ArduPilot/ardupilot #8637: Rover: support DO_SET_REVERSE commands in guided

rmackay9, 2018-06-14:
Rover-3.4 accepts DO_SET_REVERSE mission commands used in Auto mode but it is not possible for a companion computer or GCS to cause the vehicle to go in reverse in Guided mode.  We should add this ability to accept the DO_SET_REVERSE command received within a COMMAND_LONG or COMMAND_INT message.

This change involves:

- modifying the APMrover2/GCS_MAVLink.cpp file so that it accepts the DO_SET_REVERSE command and then calls Mode::set_reverse (which doesn't exist yet)
- move the ModeAuto::set_reverse() method and _reversed member up to the Mode class (i.e. promote them up one level).  In this way all modes can potentially implement reversing (but only autonomous modes like RTL, Guided, SmartRTL should try to do this, manual modes should not implement this function).
- modify ModeAuto, ModeRTL and ModeSmartRTL to pass the _reversed member into the steering controller like Auto already does.

By the way, the DO_CHANGE_SPEED command (also received within a COMMAND_LONG or COMMAND_INT message) allows the user to set the speed of the vehicle.  One idea is to accept a negative speed as meaning the vehicle should reverse.  My feeling though is that a negative speed should be interpreted as the user wants the vehicle to back away from the target waypoint not turn around and reverse towards the target.

gmorph, 2018-06-14:
I think we should have DO_CHANGE_SPEED with a negative number have the rover drive in reverse at that speed.  I have users using it that way right now (they slow down in reverse for a rough patch of ground).  I think we should have a different mav command for backing away from a WP.  That's a very specific behaviour  and whilst not useful for Planes copters and ROVs might need it.

rmackay9, 2018-08-22:
This is resolved in master and will go out with Rover-3.5.

closed 2018-08-22 by rmackay9